mjg59/python-broadlink #705: New release with support for RM4C mini (product ID 0x520D)?

mt0321, 2022-08-02:
Hello,

Could we get a new release supporting the Bestcon RM4C mini with product ID 0x520D? 

Support was [merged in to the new_product_ids branch](https://github.com/mjg59/python-broadlink/pull/694) in June thanks to [titusthefox04](https://github.com/titusthefox04), but it's [not in the main branch](https://github.com/mjg59/python-broadlink/compare/master...new_product_ids) and also [not available via Home Assistant](https://community.home-assistant.io/t/broadlink-rm4c-mini-not-supported/439216/9), presumably because the last release was [0.18.2 in May](https://github.com/mjg59/python-broadlink/releases/tag/0.18.2).

Let me know if I can help. Thank you!

seb5594, 2022-08-08:
Would be very nice if this device could get be added into the main branch :)

guieiras, 2022-08-08:
Can't wait for this release to. I bought 3 RM4C mini with product ID 0x520D and would love to have native support.
---

Tip: If someone wants to support new Product IDs right now, [patching manually the Home Assistant instance](https://github.com/mjg59/python-broadlink/pull/476#issuecomment-741993771) can be a solution.

1) Open Home Assistant and install "SSH & Web Terminal" add-on
2) Disable Protection mode
3) Connect to the Home Assistant instance via SSH:

```bash
ssh USER@IPADDRESS
docker exec -it homeassistant /bin/bash
cd /./usr/local/lib/python3.10/site-packages/broadlink
wget -O __init__.py https://raw.githubusercontent.com/mjg59/python-broadlink/new_product_ids/broadlink/__init__.py
```

4) Restart the server.

OGB1952, 2022-08-10:
I'm another one that has just bought a few of these newer RM4C Minis and have run into the same issue, so if there is an easy way to patch HA to recognise these product ID 0x520D RM4C Minis I would be grateful for any tips (I am definitely a novice!).

guieiras, 2022-08-10:
@OGB1952 You can try the steps on my comment above. After step 4 (restart the server), your HA should be able to recognize the RM4C using the native Broadlink integration.

OGB1952, 2022-08-11:
Many thanks, but try as I might I cannot find the location of the broadlink directory, or even the location of the Python directory.  I'm running HA on a Raspberry Pi using the standard HAOS installation, so not using Docker.  I've hunted high and low through the file system to try and locate where the file containing the definitions may be lurking to no avail.   This is all that is in the /usr/local/lib directory:

```
[core-ssh /]$ cd /./usr/local/lib
[core-ssh lib]$ ls
libjemalloc.a      libjemalloc.so     libjemalloc.so.2   libjemalloc_pic.a  pkgconfig

```

It's a bit frustrating, as if I've understood things correctly it looks like all the needs to be done is to add a line in a file to define the 0x520D product ID.

iron-2, 2022-08-22:
Just got mine RM4C mini and have the same issue. Hopefully this is fixed soon.

seb5594, 2022-08-26:
> Can't wait for this release to. I bought 3 RM4C mini with product ID 0x520D and would love to have native support.
> 
> ---
> 
> Tip: If someone wants to support new Product IDs right now, [patching manually the Home Assistant instance](https://github.com/mjg59/python-broadlink/pull/476#issuecomment-741993771) can be a solution.
> 
> 1) Open Home Assistant and install "SSH & Web Terminal" add-on
> 2) Disable Protection mode
> 3) Connect to the Home Assistant instance using SSH:
> 
> ```bash
> ssh USER@IPADDRESS
> docker exec -it homeassistant /bin/bash
> cd /./usr/local/lib/python3.10/site-packages/broadlink
> wget -O __init__.py https://raw.githubusercontent.com/mjg59/python-broadlink/new_product_ids/broadlink/__init__.py
> ```
> 
> 4) Restart the server.

This does not work anymore since latest HassOS update. The python packages seems to have a new location. Do you know the new location?

OGB1952, 2022-08-26:
I've spent a fair bit of time trying to find where the packages have moved to with no luck, I'm afraid.  It's irritating knowing that a simple file edit will probably fix this, but not being able to find where that file now is.

iron-2, 2022-08-26:
Where can I leave a post so that the devs get a message on this issue??

guieiras, 2022-08-26:
I feel sorry that you couldn't find the files with the guide on my comments above, but I’ll try to be more specific and I hope you should be able to manually patch your instance.

1) Use the Add-on from *Home Assistant Community Add-ons*, not from *Official Add-ons*
[image]

2) It’s very important that you disable **Protection Mode**
3) Connect to your instance using SSH. That’s what you should see when instance is connected
[image]
4) Follow the steps 3-4, the “docker” command should work no matter what is your installation.

KennyHec, 2022-09-14:
For those having problems with python dir not being found, use this instead in the terminal. This has always worked for me. 

```
docker exec -it homeassistant /bin/bash
pip3 install git+https://github.com/mjg59/python-broadlink.git@new_product_ids --force-reinstall --no-deps
```

FlameSoulis, 2022-09-26:
I'm reporting that the above instructions are incorrect. I have a core installation on a virtual machine, and I'm noticing that python is entirely missing. Perhaps this is the cause of all the problems?

In addition, the docker command doesn't work either because the command doesn't exist.

[image]

domoaligato, 2022-10-12:
the workaround works for me on unraid in docker.

simonbrunton, 2022-10-21:
RM4C mini is still not supported in 2022.10.5

ALEXOTANO, 2022-10-30:
I can't believe this haven't been solved, it is already merged to a branch and super-tested by the community, you should merge it already to main branch.

RandyLevensalor, 2022-11-02:
@felipediel @mjg59 Is there anything we can do to help get a new release published?  This patch has been merged and it would be great to have it included it included without needing to mod home assistant.

lobbie1978, 2022-11-07:
Same here. Command line does not work: "command not found".
I have been waiting for support for this device for months now. Please merge the branch to main asap!

KennyHec, 2022-11-07:
> Same here. Command line does not work: "command not found".
> I have been waiting for support for this device for months now. Please merge the branch to main asap!

What command are u (trying to) using?

bisquit2003, 2022-11-08:
> Same here. Command line does not work: "command not found". I have been waiting for support for this device for months now. Please merge the branch to main asap!

check addon, you need WEB terminal, not terminal

Druidblack, 2022-11-13:
The above method no longer works.

`docker exec -it homeassistant /bin/bash
pip3 install git+https://github.com/mjg59/python-broadlink.git@new_product_ids --force-reinstall --no-deps`
P.S. When will the normal support for these remotes be added?

closed 2022-11-20 by felipediel