mjg59/python-broadlink #706: Catching ANY Broadlink Exception with a Base Exception

bytos, 2022-08-03:
I don't understand the code 100% and don't know what exceptions are possible for each these functions offered for the RM4 Mini remote: hello(), auth(), check_temperature(), check_humidity(), send(), learn().

That's why I'd like to simply catch any of the possible exceptions in a generic way and print a message.

Is this possible? What are the possible exceptions for each of the 6 functions listed above?

felipediel, 2022-08-05:
You can do it with BroadlinkException. Here's a list of all exceptions https://github.com/mjg59/python-broadlink/blob/a86e9cbb9ce44c25485226d46541a1cdcff940b7/broadlink/exceptions.py#L114-L136.

closed 2022-08-05 by felipediel

bytos, 2022-08-05:
To avoid debugging the library, I'd have to test for all/some each time I try to blindly catch any exceptions.

Is there a document where it is indicated which exceptions can be raised by each function in the library (suspicion is no from older posts)?

You are supposed to catch "expected" exceptions, but they can't be expected if you don't know which exceptions are raised by a function.

Ended up catching any exceptions (not using BL exceptions) because it is shorter.

felipediel, 2022-08-05:
Most of them come from the firmware. There is no manual, at least we don't have access to it. In Home Assistant we usually catch BroadlinkException and use specific exceptions in known cases.

bytos, 2022-08-05:
Thanks for the info. My application works standalone and is responsible for turning off equipment, hence, the need to know the options.