mjg59/python-broadlink #712: python-broadlink supports LEDvance APM ceiling light (kind of)

someonefromsonewhere, 2022-08-15:
This is not exactly an issue, but I've been tinkering and noticed that python-broadlink supports LEDvance APM ceiling light with slight modifications.

Important Note :
This works only for LEDvance 4SMART+ WiFi Ceiling TW 24W (SKU : 4058075484887) which uses the "LEDVANCE **APM** SMART+ Wi-Fi" App and is based on Broadlink. There are a few other bulbs which uses the "LEDVANCE **APM** SMART+ Wi-Fi" app that I suspect will work but I do not have them to test. It **will not work** on lights/bulbs which uses the "LEDVANCE SMART+ Wi-Fi" and is based on Tuya platform.

These ceiling light has device code of 0x6498. Adding this device code to lb1 allows the light to be added into home assistant. Power on/off, changing brightness works flawlessly. Changing colour temperature works but it is reversed (i.e. increasing temperature in home assistant results in the bulb getting warmer and vice versa). Tried setting it to lb2 but it won't work when adding device.

Here are the steps I took.

1. Add 0x6498 to lb1 (credit to https://community.home-assistant.io/t/broadlink-rm4c-modify-python-library/258914/4)
- Install and open SSH & Web Terminal addon
- type docker exec -it homeassistant /bin/bash
- Enter the command: cd /./usr/local/lib/python3.8/site-packages/broadlink
- Open the file using the command: vi init.py
- Inside the file Type “i” (This will start edition mode on the file).
- Under the section “Supported Types”, "lb1" add the line 0x6498: (LB1, “Broadlink”),
- Press ESC
- Type “:wq” and then press enter.
- Reboot HASSIO.

2. Join light to wifi
- reset light by turning on and off 5 times
- Connect PC to light's AP "LEDVANCE_Smart"
- launch python and import broadlink
- call broadlink.setup('WIFI_SSID', 'WIFI_Passcode', 3)
- At this point, PC should drop from the "LEDVANCE_Smart" AP

3. Configure light using python-broadlink 
- At this point, PC should join back to your regular wifi ssid.
- In Python do this
     - device = broadlink.hello('IP_of_Light") 
     - device.auth()
     - device.set_lock(False)

4. After the above is done, I was able to add a broadlink device using the IP of the light.

I suspect this may break when there is an update to python-broadlink in home-assistant.

Would appreciate if contributor(s) to this code can add 0x6498 into the supported device. (Bonus if the inverse colour temperature could be looked into. I would like to contribute but I'm not sure how to. I am happy to do any further testing with this type of light.

PenguinBlock, 2023-09-01:
Thank you so much for sharing your information.
I am now able to add my LEDVANCE ceiling lights to HA.

The device code of my light is 0xA7B6 and it works fine when set to lb2.
However, the color temperature change works in reverse.
I'll have to look into this a bit more.

lunDreame, 2024-05-01:
> 정보를 공유해 주셔서 정말 감사합니다. 이제 LEDVANCE 천장 조명을 HA에 추가할 수 있습니다.
> 
> 내 조명의 장치 코드는 0xA7B6이며 lb2로 설정하면 제대로 작동합니다. 그러나 색온도 변화는 반대로 작용합니다. 이건 좀 더 알아봐야겠네요.

Can you explain in more detail? I have the same lighting ID as you.

Do I just need to add a new light id to the __init___.py of the Home Assistent broadlink package? As shown in the guide above, after installing the broadlink package on the local computer, adding the light id, it is not possible to connect through the package in the AP mode of the initialized device. (Device was not found) Eventually, after linking through the existing leadance AIPM app, it succeeded in importing information from the device through the lighting IP, but the error occurs in the auth (broadlink.exception.AuthenticationError: [Errno-1] Authentication failed).

PenguinBlock, 2024-05-01:
> 
> Can you explain in more detail? I have the same lighting ID as you.
> 
> Do I just need to add a new light id to the **init**_.py of the Home Assistent broadlink package? As shown in the guide above, after installing the broadlink package on the local computer, adding the light id, it is not possible to connect through the package in the AP mode of the initialized device. (Device was not found) Eventually, after linking through the existing leadance AIPM app, it succeeded in importing information from the device through the lighting IP, but the error occurs in the auth (broadlink.exception.AuthenticationError: [Errno-1] Authentication failed).
> 
> To unlock the device CW (192.168.1.17 LB2) is locked. The instrument must be unlocked to authenticate and complete the configuration. Follow these steps:
> 
> Please open the Broadlink app. Please click on the device. Click ... at the top right. Please scroll to the bottom of the page. Please unlock it. invalid_auth

한국 분이신 것 같아서, 여기 예의가 아닐 수도 있지만 한국어로 말씀드릴게요!
컴퓨터에서 파이썬을 직접 실행해서 step 2,3 을 진행하시면
앱 없이도 인증과 함께 홈 네트워크에 연결하고, lock 상태를 풀 수 있어요.
이 이후에 HA 에 broadlink 패키지를 설치하고, 터미널로 init.py 파일을 수정해서 ID 를 넣으시면 됩니다.
문제 없이 진행됐다면 broadlink 구성요소에서 IP 를 통해 조명을 추가하실 수 있을꺼에요.

lunDreame, 2024-05-01:
> 한국 분이신 것 같아서, 여기 예의가 아닐 수도 있지만 한국어로 말씀드릴게요!
>
> 컴퓨터에서 파이썬을 직접 실행해서 step 2,3 을 진행하시면
>
> 앱 없이도 인증과 함께 홈 네트워크에 연결하고, lock 상태를 풀 수 있어요.
>
> 이 이후에 HA 에 broadlink 패키지를 설치하고, 터미널로 init.py 파일을 수정해서 ID 를 넣으시면 됩니다. [...]

성공했습니다. ip를 잘못 잡고 있었네요.. 로컬로 연동하고 신세계를 경험하고 있는 중입니다 ㅎ