mjg59/python-broadlink #719: Luceco Castra (Hangzhou Broadlink) discoverable, but gives Authentication Error

rbubley, 2022-09-25:
`python-broadink` can discover a Luceco Castra smart light with PIR (reporting `devtype=25722, name='ACpir'`), but attempting to `auth()` the device throws an exception:
```
>>> import broadlink
>>> dev=broadlink.hello('192.168.XXX.XXX')
>>> dev
broadlink.device.Device(('192.168.XXX.XXX', 80), mac=b'\xXX\xXX\xXX\xXX\xXX', devtype=25722, timeout=10, name='ACpir', model='', manufacturer='', is_locked=True)
>>> dev.auth()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/russ/.local/lib/python3.10/site-packages/broadlink/device.py", line 185, in auth
    e.check_error(response[0x22:0x24])
  File "/home/russ/.local/lib/python3.10/site-packages/broadlink/exceptions.py", line 152, in check_error
    raise exception(error_code)
broadlink.exceptions.AuthenticationError: [Errno -1] Authentication failed

```

zhou9110, 2022-09-30:
I found the solution in here, it is because `is_locked=True`.

In my case I am using RM4 pro, I need to open the BroadLink App, select the device -> click "..." in the top bar -> click "Property", then find the "Lock device" setting and turn it off. Then calling `device.auth()` just works for me.

Reference: https://github.com/mjg59/python-broadlink/issues/674#issuecomment-1134248621

felipediel, 2024-04-21:
Let's add it. Does it work with the `lb1` class?

echokk11, 2024-12-04:
Unfortunately, there is no is_locked column in the gateway property settings of Hangzhou Broadlink App which named "博联云想"