mjg59/python-broadlink #721: When send same command for second time, the RM Pro + sends the command multiple times

vdleun, 2022-11-05:
I have learned Home Assistant installation the IR codes of the Amino Aria 710 settop box and also through the cli tool broadlink_cli.py. I'm using a Broadlink RM Pro +. When sending the command for example for channel up just once, the stb goes correctly one channel up. When sending the same command again, the Broadlink seems to send the IR codes multiple times, because the stb goes up a couple of channels. This problem exists when sending the IR code with Home Assistant and when using the cli tool broadlink_cli.py.

HA: 2022.11.1
Supervisor: 2022.10.2
Operating System: 9.3

python-broadlink was installed through pip3.

A snippet of the config in HA:

`"tvbox": {
      "kanaal-omhoog": "JgDGAFcdEA4PDRAcDx0rHRANEA0QDg8ODw4PDh4cDw4QDRANHg4PDg8cEA0fGxAOHg0QHA8ODw4eHA8OEA0QDRANEAAI0FgdDw4PDhAcDxwrHg8ODw4QDRANEA0QDR4cEA0QDRAOHg0QDRAcDw4eHBANHg4PHBANEA0fGxAODw4PDg8ODwAI0VgcEA0QDRAcDx0rHRANEA0QDRAODw4PDh4cDw4PDhANHg4PDg8cEA0fGxAOHg0QHA8ODw4eHA8OEA0QDRANEAANBQAA",
      "kanaal-omlaag": "JgDAAFcdEA0QDRAcDxwtHBANEA0QDRANEA0QDR8bEA4PDg8OHg0QDRANEBweHBANHg4PHBANEA4eHA8ODw4PDh4ACN9YHBAODw4PHBAcLBwQDg8ODw4PDg8ODw4eHBANEA0QDR4ODw4PDhAbHxwPDh4NEBwPDhANHhwQDRANEA0fAAjfVx0QDRANEBwPHC0cDw4QDRANEA0QDRANHxsQDg8ODw4eDRANEA0QHB4cEA0eDg8cEA0QDh4cDw4PDg8OHgANBQAAAAAAAAAA"
    }`

The same IR codes when saved by broadlink_cli.py in a file:

tvbox-channelup:
`2600c600581c0f0e100d101c0f1d2a1e0e0f100d100e0f0e0e0f0f0e1c1e0f0e0f0e0f0e1e0e0f0e0f1c0e0f1e1d0e0f1e0d101c0f0e0f0e1c1e0e0f0f0e100d100d100008d0581d0e0f0f0e0f1d0e1d2a1f0e0f0f0e0e0f0f0e100d0f0e1e1c0e0f100d0f0f1b100f0e101c0e0f1e1c0e0f1e0e0e1d0f0e100e1c1e0e0f0f0e0f0e0f0e0f0008d1581c0e0f100d101c0e1e2a1e0e0f100d100d100e0e0f0f0e1e1c0e0f0f0e0f0e1e0e0f0e0f1c0f0e1e1c0f0f1e0d0f1d0e0f0e0f1e1c0e0f0f0e0f0e0f0e0f000d050000`

tvbox-channeldown:
`2600c000581c0f0e100d0f1d0f1c2e1b0f0e0e0f0e0f100d100d0f0e1f1c0f0e0f0e100d1e0d0f0e110c0f1d1e1c0f0e1f0d0f1c100d100e1d1d100d0f0e0f0e1d0008e0581c0f0f0e0f101b0f1d2b1d110d100d0d100d100f0e100d1d1d0f0e0f0e100d1e0e0f0e0f0e0f1d1c1e0f0e1d0e101c0f0e0e0f1c1e0f0e100d0f0e1e0008e0581c0f0e0e100f1c0f1c2c1d0f0e100d100d100d100d100d1f1c0f0e100d0f0e1e0d0f0e110c101c1d1d0f0e1e0e0f1c100d0f0f1e1c0f0e100d0e0f1d000d050000000000000000`

Please let me know if I can do anything to solve or debug this problem.

RestOp, 2023-01-12:
The same by me. And then after some time RM4 Pro reboots

brent20, 2023-07-03:
Same issue here.

denis-stepanov, 2024-03-30:
I also have this problem. RM4 Pro.