mjg59/python-broadlink #739: Learning RF doesn't work, IR does

zupancicmarko, 2023-01-10:
Hi, I'm using Home Assistant and trying to get Broadlink to learn RF from Philips TV remote control. No success in HA or using CLI here.

This IR command works well:
`python broadlink_cli --type 0x279d --host ************ --mac ************ --learn`
 
Where in response I get:
`Learning...
26008400581d0f0f0d110c1e0f1d2b1e0f0f0e0f0d100d110c120d0f1b1f0e0f0d100e101b100f0e0e1e0e0f1c1f0e0f1c100e1d0f0f0e0f1c100d1e100e0d110c100e0008e9561e0f0f0e0f0d1f0e1d2b1e0f100c100e0f0e100c100e101c1e0e0f0e0f0f0e1c100e100e1d0f0e1c1f0e0f1c100e1d0f0f0e101b100e1d0f0f0e100c110c000d0500000000
Base64: b'JgCEAFgdDw8NEQweDx0rHg8PDg8NEA0RDBINDxsfDg8NEA4QGxAPDg4eDg8cHw4PHBAOHQ8PDg8cEA0eEA4NEQwQDgAI6VYeDw8ODw0fDh0rHg8QDBAODw4QDBAOEBweDg8ODw8OHBAOEA4dDw4cHw4PHBAOHQ8PDhAbEA4dDw8OEAwRDAANBQAAAAA='`

However learning the RF code simply doesn't work. The RF command:
`python broadlink_cli --type 0x279d --host ************ --mac ************ --rfscanlearn`

Response:
`Learning RF Frequency, press and hold the button to learn...
RF Frequency not found`

I tried to hold down the button, tried pressing the button in 1sec sequence. Tried different button. No luck.

Any suggestions? Thank you.

[image]

bengtmartensson, 2023-01-13:
The "RF" the Broadlink understands are some particular protocols on 433 and 315  MHz. There are other protocols, and it looks as if your Philips uses one such. You may have success using e.g. a Flipper Zero.

zupancicmarko, 2023-01-13:
> The "RF" the Broadlink understands are some particular protocols on 433 and 315 MHz. There are other protocols, and it looks as if your Philips uses one such. You may have success using e.g. a Flipper Zero.

Thanks for the feedback. Probably gonna use that money and do the TV upgrade. 😉

romelec, 2023-12-24:
It's a known issue with a fix available, by specifying the frequency to scan, but unfortunately never added: #613

closed 2024-04-21 by felipediel