mjg59/python-broadlink #776: Error in line 7, in <module>     import broadlink

daserra23, 2024-01-17:
Hi, I am trying to migrate from Buster to Bullseye on a raspberry pi. I use the broadlink_cli with scripts.

I have installed broadlink

`admin@domoticz2:~ $ pip3 install broadlink                                      Defaulting to user installation because normal site-packages is not writeable
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: broadlink in ./.local/lib/python3.9/site-packages (0.18.3)
Requirement already satisfied: cryptography>=3.2 in /usr/lib/python3/dist-packages (from broadlink) (3.3.2)
DEPRECATION: gpg 1.14.0-unknown has a non-standard version number. pip 24.0 will enforce this behaviour change. A possible replacement is to upgrade to a newer version of gpg or contact the author to suggest that they release a version with a conforming version number. Discussion can be found at https://github.com/pypa/pip/issues/12063`

But when I run this script/command

`sudo /home/admin/python-broadlink/cli/broadlink_cli --host 192.168.2.20 --mac 24dfa7d60a60 --type 0x5f36 --send 260040000001108b133412121212121213341212121214101236113612121212111311131112121212121212121212121212113612121236111213111410123611000d05`

I get the error above at line 7 of the broadlink_cli script

Also

`admin@domoticz2:~ $ python -V
Python 3.9.2
`
Any ideas what I am doing wrong ?

Nardol, 2024-01-17:
Except if there is something specific to your installation, you should be able to execute broadlink-cli without sudo privileges and it should work.

But if you really have to use sudo to execute broadlink-cli, install python-broadlink by prefixing pip by sudo.
It will install python-broadlink in system packages.

daserra23, 2024-01-17:
OK, very basic error of mine, should have been sudo pip3 install broadlink

daserra23, 2024-01-17:
> Except if there is something specific to your installation, you should be able to execute broadlink-cli without sudo privileges and it should work.
>
> But if you really have to use sudo to execute [...]

Thanks for your super rapid response, all working now.

closed 2024-04-09 by felipediel