mjg59/python-broadlink #777: Can we get a new release?

hungyao, 2024-02-10:
Hi there - would it be possible to create a new release with recent pull requests incorporated?  The last release was over a year ago, so I think it means that new devices are not being supported in upstream projects that are using this library such as Home Assistant, etc.

hungyao, 2024-03-14:
This project is looking somewhat abandoned... would it be prudent for someone to fork it and create a release that Home Assistant can reference?

MichielioZ, 2024-04-03:
> Can you merge these pulls and create a new release, @felipediel @mjg59 ? Thanks.

Sorry to be the bearer of bad news, but as far as I know both @mjg59 and @felipediel are no longer active on this repo.
For a while it looked like @felipediel was going to pick up maintaining (and even enhance) the code, but he was and is (?) working on this code for Home Assistant. However, they (=people at Home assistant) have forked and altered this code to fit their project and it doesn't look like any of the updated/enhanced code made it in here :-(

Long story short: there are people still working on/maintaining  Broadlink Python code, but unless you use Home Assistant (or port changes over to a fork yourself) it's not easily available anymore ...

felipediel, 2024-04-03:
Hi friends. It's been challenging to balance open source projects with my work and other responsibilities. I still need to find the right balance between all of this. That said, this project is not abandoned, as soon as I find time I will create a new release.

closed 2024-04-21 by felipediel