mjg59/python-broadlink #778: Sending duration (IR-style) RF transmissions

eXeC64, 2024-02-26:
protocol.md mentions that the format for sending signals with broadlink for IR is pairs of on/off durations, however for RF this does not seem to be the case. With my RM4 Pro and observing with a rtl-sdr dongle, it appears that instead of being on/off durations, the data sent is just the delay between momentary on transmissions. So I can control how long in between pulses, but not how long each pulse is.

This kind of signalling is insufficient for the blinds in my house (Sunbell) which require a series of pulses of different lengths to be transmitted over RF. I've tried different modes (changing b2 to nearby values changes the RF frequency, but not the signal itself), but I've not been able to get broadlink to do something like this.

Has anyone been able to do on/off/on/off duration style transmissions over RF? If so, how? There's clearly bits of the protocol not fully reversed, as the alternate RF values of b0, b1, b3, etc. transmit on nearby frequencies but aren't documented anywhere that I can find online.

clockbrain, 2024-02-28:
I only use my Broadlink for RF and it certainly does handle variable length on and off pulses. The Broadlink doesn't actually understand the transmitted protocol, it just sends those variable length pulses as instructed by the recorded sequence.

Have you managed to record an RF sequence with the Broadlink and play it back?

There is more detail on the broadlink file protocol in this comment thread, particularly towards the end [https://github.com/mjg59/python-broadlink/issues/57](https://github.com/mjg59/python-broadlink/issues/57)

eXeC64, 2024-03-11:
> Have you managed to record an RF sequence with the Broadlink and play it back?
No, it always fails the record the signal I'm trying to capture.

In testing I've resorted to generating a series of 3 repeating values with a ratio of 1000:500:250, repeated 30 times. If this were variable length on/off pulses, I'd expect to see that when recording. But what I record looks like the following, which seems to be a series of momentary peaks separated by gaps of ratio 1000:500:250.

[image]
Message:
`b2 3 5a 0 1e f 8 1e f 8 1e f 8 1e f 8 1e f 8 1e f 8 1e f 8 1e f 8 1e f 8 1e f 8 1e f 8 1e f 8 1e f 8 1e f 8 1e f 8 1e f 8 1e f 8 1e f 8 1e f 8 1e f 8 1e f 8 1e f 8 1e f 8 1e f 8 1e f 8 1e f 8 1e f 8 1e f 8 1e f 8 1e f 8`

Presumably I'm doing something wrong, but I'm really struggling to figure out what. Nothing I've generated has ever resulted in a non-momentary pulse from broadlink.

clockbrain, 2024-03-12:
That's a shame you can't record the original signal. Makes it much harder to progress.

Have you tried the specified frequency version of the code, i.e. [https://github.com/brentleeper/python-broadlink](https://github.com/brentleeper/python-broadlink)? I found it much easier to record my signals with that patch.

As an example of an RF signal that does work, here is a short one I use to set my ceiling fan to speed 1.

> b1c04400009f06000d950d230d23250b250b0d23250b250b0d23250b0d23250b0d230d23250b0d23250b0d23250b0d23250b250b250b250b250b0d23250b0d230d230d230d9505dc

I send this signal as a text file to the Broadlink as follows:

> ./broadlink_cli --device @RM4PRO.device --send @test_codes.txt

Here is a breakdown of how that signal is structured so Broadlink can process it:

> b1c0: RF 433Mhz
4400: little endian body length = 00*256 + 44 = 68 bytes
009f: unknown
0600: unknown
0d95: 0d high then 95 low (pulse lengths) = initial gap
0d23: 0d high then 23 low = 0
0d23: 0d high then 23 low = 0
250b: 25 high then 0b low = 1
250b: etc
0d23
250b
250b
...
...
...
250b
0d23
0d23
0d23
0d95
05dc: end??

This is what a chart of those pulse lengths looks like which can easily be read as 1's and 0's
[image]

This is a capture from URH of my Broadlink sending that same signal so you can see the correlation of the pulse lengths.
[image]

I don't know how you have generated that message you are attempting to send but if it isn't in the right structure the Broadlink won't be able to process it to transmit. It needs those other bits such as payload length and termination bytes.