mjg59/python-broadlink #800: get_subdevices with RM4Pro

eadmaster, 2024-04-20:
is it possible to read configured subdevices in the app with the RM4Pro?

I've tried with `device.get_subdevices()`, but the returned list is empty.

If possible this could bypass the need to [backup the learned codes from the app](https://github.com/mjg59/python-broadlink/issues/559).

felipediel, 2024-04-20:
The term "subdevices" can be confusing because it has different meanings in this API and in the official Broadlink app. In the context of this API, "subdevices" refers to Broadlink devices connected to a master Broadlink device through a separate network interface. In the official Broadlink app, "subdevices" are virtual representations of IR/RF devices and their associated codes, which are stored in the cloud. That said, this API cannot access, manage, or interact with the cloud-stored "subdevices" from the Broadlink app.

closed 2024-04-20 by felipediel

eadmaster, 2024-04-20:
> That said, this API cannot access, manage, or interact with the cloud-stored "subdevices" from the Broadlink app.

Do you know if there is a tool that can do that? the latest official Remote Control app has a "backup" feature, and it is also caching some data in `/storage/emulated/0/Android/data` (i'm trying to investigate these files if they contain something useful)

iArtuc, 2024-11-07:
@eadmaster i'm curious if you solved the issue, i have the same issue, and trying to set the device state but failing