mjg59/python-broadlink #804: RM Mini 3 Not Connecting

Jab2870, 2024-06-10:
Hi

I've recently got a new RM mini 3. Following the instructions in the readme, I connect to it's access point.

However, the setup function doesn't appear to do anything.  No errors, but also no output. 

```python
>>> import broadlink
>>> broadlink.setup('mysmartnetwork','password',3, ip_address='192.168.10.255')
```

However, the device never appears to connect to my network - nothing in my router anyway. Not really sure where to go next in terms of debugging. 

Thanks in advance

Nardol, 2024-06-10:
Is your computer only connected to the Broadlink access point so not using E.G. Ethernet?

Nardol, 2024-06-10:
Or did you try to specify a broadcast address E.G. `broadlink.setup('myssid', 'mynetworkpass', 3, ip_address='192.168.0.255')`

Jab2870, 2024-06-10:
Thanks for the prompt reply.

> Is your computer only connected to the Broadlink access point so not using E.G. Ethernet?

Yes, only connected to the Broadlink device, although re-connects to my home network once the device's AP goes down.

> Or did you try to specify a broadcast address E.G. `broadlink.setup('myssid', 'mynetworkpass', 3, ip_address='192.168.0.255')`

Yes, the broadcast address for the network I am trying to connect it to is 10.255, but I did try that.

Jab2870, 2024-06-10:
And just to add, even if the broadcast was failing, I'd spill expect to see the device connected to the network after authenticating.

eurodrew555, 2024-12-29:
I had this same issue trying to connect my RM Mini 3 using an existing 2.4 Ghz SSID  
      
I managed to solve this by creating a new SSID using Legacy mode on 2.4Ghz just for the RM Mini 3 to connect to.  
  
My WAPs are running OpenWRT, examples for SSID and Key:  
  
SSID = EXAMPLE-SSID-JUST-FOR-RM-MINI3
Encryption mode = WPA-PSK/WPA2-PSK Mixed Mode  
Cipher = auto  
Key = mysupersecretkey
  
Put the RM Mini 3 into AP mode and connected to it using my iPhone   
This network is 192.168.10.x so you need to use 192.168.10.255 as the broadcast address below for this to work.  
I used the iOS App a-shell - https://holzschu.github.io/a-Shell_iOS/  to run the following python code:  
  
```
pip install broadlink

python
import broadlink
broadlink.setup('EXAMPLE-SSID-JUST-FOR-RM-MINI3', 'mysupersecretkey', 4, ip_address='192.168.10.255')
```
  
The RM Mini 3 then successfully joined my network.