mjg59/python-broadlink #808: Feature Request: Enable specifying the frequency for learning

CitizenRacer, 2024-07-18:
The broadlink app allows you to set the frequency to listen to for learning a code. This is the only way I can get it to learn my fireplace remote, the sweep function doesn't discover the remote. I'd like to be able to specify the frequency in the library so that I can skip the sweep and have the app listen for the specific frequency and find my codes.

gorbypark, 2024-09-25:
I think this is a great solution.  The Broadlink app finds all my RF codes almost instantly, but using Home Assistant or Broadlink Manager (both using this project under the hood), the sweep frequency learning is very unreliable.  Sometimes it works, sometimes it doesn't, and most times it takes holding the button 10-20 times before it picks up anything.

CitizenRacer, 2024-10-01:
> Was surprised to see this is not a feature. Is it supported by the broadlink device? Do we just need to add it to this library?

It is supported by the BroadLink device.

MitchTalmadge, 2024-10-01:
Calling `device.find_rf_packet(315.0)` worked great for me with a 315MHz fireplace remote.