mjg59/python-broadlink #815: Long Wifi Password results in "IndexError: bytearray index out of range"

Jamolo5, 2024-11-19:
When trying to set up a device, I run the command:
`broadlink.setup('mywifi','some37characterlongpassword',3,ip_address='192.168.1.255')`
Which results in
```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/jamo/.local/lib/python3.10/site-packages/broadlink/__init__.py", line 318, in setup
    payload[(pass_start + pass_length)] = ord(letter)
IndexError: bytearray index out of range
```

The original reason I was wanting to use this package is because the Broadlink app doesn't support my wifi password length

ChrisAcrobat, 2025-02-02:
I'm having the same problem and with the same reason.

ChrisAcrobat, 2025-02-02:
The device has a 32 character limit. #235 👎