mjg59/python-broadlink #824: Authentication Error With New RM4 Mini

bytos, 2025-03-30:
After using direct discovery with a new RM4 Mini, can't authenticate with the "auth()" method:

`Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "d:\Apps\test\.venv\Lib\site-packages\broadlink\device.py", line 185, in auth
    e.check_error(response[0x22:0x24])
  File "d:\Apps\test\.venv\Lib\site-packages\broadlink\exceptions.py", line 152, in check_error
    raise exception(error_code)
broadlink.exceptions.AuthenticationError: [Errno -1] Authentication failed`

smithlevenson, 2025-03-31:
I am getting the same error on a RM4 Pro after it is discovered:

`[broadlink.remote.rm4pro(('192.168.1.XX', 80), mac=b'\xe8pr\x9e\xadz', devtype=21003, timeout=10, name='', model='RM4 pro', manufacturer='Broadlink', is_locked=True)]`

bytos, 2025-03-31:
At first, the problem was the AP not allowing the remote to connect, once fixed, couldn't get auth() to work and the app never finished the setup with the 3rd step giving an error.

Tried to setup with the app several times without apparent success, but the auth() method is working now.

I think I used the app years ago with the same model to unlock it, maybe it was unlocked while trying to set it up with the app, even though it always gave that error.

smithlevenson, 2025-03-31:
So does the device need to be unlocked in order for .Auth() to work?

> At first, the problem was the AP not allowing the remote to connect, once fixed, couldn't get auth() to work and the app never finished the setup with the 3rd step giving an error.
> 
> Tried to setup with the app several times without apparent success, but the auth() method is working now.
> 
> I think I used the app years ago with the same model to unlock it, maybe it was unlocked while trying to setup with the app, even though it always gave that error.

bytos, 2025-04-01:
When using setup(), I must use ip_address="192.168.0.255" according to the instructions in the Github main page.

Broadlink device unlock #449
https://github.com/mjg59/python-broadlink/issues/449

broadlink.setup fails because broadcasting the data fails #618
https://github.com/mjg59/python-broadlink/issues/618

Broadlink RM4 Pro wifi setup without using the App?
https://www.reddit.com/r/homeassistant/comments/16wh0tt/broadlink_rm4_pro_wifi_setup_without_using_the_app/