mjg59/python-broadlink #829: Broadlink RM MAX is not supported

danpeled7, 2025-08-04:
I purchased a new Broadlink RM MAX which is the successor of RM4 Pro.
however it doesnt pair to the integration in home assistant - i get a message : device not supported.
can you please add support for this device? thanks!

device documentation on broadlink website:
https://ebroadlink.com/products/broadlink-rm-max-smart-home-hub

device information from broadlink app:

{
"Device IP":"192.168.1.146@80",
"DID":"00000000000000000000348e894b6c7d",
"SDK":"2.17",
"Plug-in version":"1.6.8",
"IoT Cloud":"World",
"Routine traits":"Trigger",
"S/N code":" ",
"Data Cloud":"Other regions",
"Firmware":"V62130",
"PID":"0000000000000000000000008baf0000"
}

zloyle, 2025-08-08:
Same experience with the Broadlink RM5 Plus.

ArikShemesh, 2025-08-12:
> I purchased a new Broadlink RM MAX which is the successor of RM4 Pro. however it is not discovered by python scripts and does not appear in supported devices. in home assistant i get a message : device not supported.
> 
> can you please add support for this device? thanks!
> 
> device documentation on broadlink website: https://ebroadlink.com/products/broadlink-rm-max-smart-home-hub
> 
> device information from broadlink app:
> 
> { "Device IP":"192.168.1.146@80", "DID":"00000000000000000000348e894b6c7d", "SDK":"2.17", "Plug-in version":"1.6.8", "IoT Cloud":"World", "Routine traits":"Trigger", "S/N code":" ", "Data Cloud":"Other regions", "Firmware":"V62130", "PID":"0000000000000000000000008baf0000" }

Same for me, please add support for RM MAX

joaofreire11, 2025-10-29:
I bought this device as well without confirming it would work with home assistant integration. 😔
I just wanted to add I notice this device does not show the lock/unlock option on the Broadlink app. It can be added on the new Broadlink Magic Home app and there it has an option to "connect to a 3rd-party" where it can be set as a server or as a client but I cannot find any documentation on that...

Amstutz, 2025-12-30:
Broadlink RM MAX would be highly appreciated. Same issue here.

Biz92k, 2026-01-02:
> > Same experience with the Broadlink RM5 Plus.
> 
> added (very) basic support for rm 5 plus here: [#831](https://github.com/mjg59/python-broadlink/pull/831)

any chance to do the same also for RM MAX?

Frankj999, 2026-02-20:
Another request for pro max support

shankman333, 2026-05-17:
+1 for RM Max

amasolov, 2026-05-24:
Hi @felipediel @mjg59,

I have a Broadlink RM Max and have submitted #838 to add support for it (device type `0xAF8B`, mapped to the `rm4pro` class). IR learning and sending tested successfully on my device.

I notice this issue has been open since August 2025, and PR #831 (RM5 Plus support) has been waiting for review since October 2025. There are 20+ community members requesting RM Max support.

I'd be happy to help maintain this library going forward — reviewing PRs, adding new device types, cutting releases. Would either of you be open to adding me as a collaborator?

Thanks for all the work you've put into this project!

aliharbaji, 2026-10-03:
I still don't have the device to test PR #838, but is this issue considered closed now?

josete73, 2026-10-04:
Por aqui otro peleandose con el RM Max hasta que la IA me ha mandado aqui porque no estan actualizadas las librerias y esperando solucion